realthunder/FreeCAD #507: Outdated `DiligentEngine` fork breaks clean build

twelho, 2022-07-22:
I'm building `LinkDaily` in a clean environment (as a Flatpak) with no source cache and cannot fetch the sources successfully. It seems like `LinkDaily` relies on your [DiligentEngine](https://github.com/realthunder/DiligentEngine) fork, which hasn't been updated in quite some time. The latest commit has https://github.com/DiligentGraphics/DiligentCore/tree/fdb42e213cbbab10f4a2783e00bbaaa0564cdf07 as a submodule, and in there under `ThirdParty` there's another submodule for `SPIRV-Cross` referring to the commit https://github.com/DiligentGraphics/SPIRV-Cross/tree/f61a33c8f4bbea37ac6fb3c1710288127dbc7e15, which is gone and can't be fetched, breaking the build.

If this commit happens to be present in the cache, or if I change the `DiligentEngine` submodule of `LinkDaily` to point directly at the upstream, the build finishes without issues. Could you update your fork, or avoid the fork and use the upstream directly (perhaps at the latest release tag instead of master if you want the stability)?

Finally, thanks for your hard work, it makes FreeCAD truly great!

twelho, 2022-07-22:
I've now pinned `DiligentEngine` to upstream v2.5.2 in my [fork](https://github.com/twelho/FreeCAD/commits/LinkDaily), I can open a PR if this approach works for you. The automated Flatpak builds using that fork can be tracked [here](https://github.com/twelho/org.freecadweb.FreeCAD/actions).

realthunder, 2022-07-22:
I'll update `DiligentEngine` soon. Please note that `DiligentEngine` is only used in LinkRender branch, which is not really functioning at the moment. You don't really need to run `git submodule update` to build FreeCAD at the moment.

luzpaz, 2022-10-07:
Closing issue

closed 2022-10-07 by luzpaz