realthunder/FreeCAD #508: Pressing escape in any active command, closes FreeCAD in snap version.

czinehuba, 2022-07-22:
Pressing escape in any active command, closes FreeCAD in snap version.

Simple way to reproduce is to: create a body -> create a sketch -> select a plane; then press ESC and it exits FreeCAD. Might be an issue with Ubuntu Core. 

It did it with one or two earlier releases. 

OS: Ubuntu Core 20 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 2022.721.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.07.21-edge)
Hash: 5eb4402ab6d4f085a0636c234b3b5953c984befa
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)

realthunder, 2022-07-22:
Should be fixed in 44388bfae2e3a27bf5c64c851a8b901ec92c1de8, but I forgot to trigger the build. Anyway, just wait a few hours and try again.

MiqCG, 2022-07-23:
I confirm that, with Snap 2022-07-23-gb934759e , it is now fixed (at least for me).

Thanks

czinehuba, 2022-07-23:
That is great. It works now.
Thank you for the quick response.

closed 2022-07-23 by czinehuba