realthunder/FreeCAD #511: Error when moving shapestrings around

Bonnee, 2022-07-25:
I get this error when I'm trying to move shapestrings in a model. The issue was not present a week ago
```
11:14:29  Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Draft/draftobjects/shapestring.py", line 115, in execute
    s = self.makeFaces(char)
  File "/usr/lib/freecad/Mod/Draft/draftobjects/shapestring.py", line 131, in makeFaces
    compEdges = compEdges.connectEdgesToWires()
<class 'Part.OCCError'>: 15StdFail_NotDone BRep_API: command not done
11:14:29  127.853 <App> Document.cpp(4293): Failed to recompute front_panel_new#ShapeString003: 15StdFail_NotDone BRep_API: command not done
11:14:29  Recompute failed! Please check report view.
```
I'm running the latest commit

realthunder, 2022-07-25:
Can you please provide a file showing the problem?

Bonnee, 2022-07-26:
The last commit seems to have fixed it. Thanks!

closed 2022-07-26 by Bonnee