realthunder/FreeCAD #512: [Part Design] Revolution with sketch edge on the revolution axis fails with error "gp_Dir::Crossed() - result vector has zero norm"

pathfinder555, 2022-07-26:
I usually use a sketch edge as reference when doing a revolution, but it doesn't work in the current version anymore. Using the global axes or a datumline doesn't work either, when one of the sketches edge is in the same position as the revolution axis.

Version:

OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.704.26244 +4971 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: a87320cc64699a8e4226bad2e9b37ac3a1296ab9
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

It works fine in older versions, e.g.:

OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

It works fine in regular FreeCAD, so it seems to be an issue with this version.

realthunder, 2022-07-26:
Can you please try 0709 release?

pathfinder555, 2022-07-26:
It works in 0709, thanks!

closed 2022-07-26 by pathfinder555