realthunder/FreeCAD #515: Stack overflow after creating new Body and clicking any command (pad, etc) when empty Body is selected

scottmudge, 2022-07-27:
## Affected version:

Latest tip of **LinkDaily** branch (7.25.2022 label).

## Steps to Reproduce:

1. Create new empty document.
2. Go to Part Design Workbench
3. Create new (empty) Body.
4. Do not click anything else (keep empty Body selected/highlighted in Tree view).
5. Click Pad or any other command in the toolbar, and the application will crash due to stack overflow from infinite function recursion.

## Analysis:

The `DocumentObjectItem::getSubName()` in `Tree.cpp` function infinitely recurses. It will get to the following line, and then re-enter itself, infinitely until stack overflow:

`res = item->getSubName(str, topParent);`

## Debugger Screenshot:

[image]

Note:

* `Auto group solids in Body` is **enabled**.
* `Use unique label for auxillary groups in Body` is **disabled**.
* Auxiliary-grouping mode is active on the body.

discip, 2022-10-04:
Confirmed for 2022.709.

closed 2022-10-05 by realthunder

discip, 2022-10-05:
Perfect!
Thank you!

commit fb1d2ad5 (2022-10-05): Gui: fix crash on dependency loop

Fixes #515
@scottmudge