realthunder/FreeCAD #517: Crash when deleting constraints

pgilfernandez, 2022-07-28:
It happens everytime you want to delete a numerical constraint, FreeCAD crashes.
I'm under macOS 12.3 and it even happens with new files:

1. open a new file
2. create a new sketch
3. draw something
4. apply a numerical constraint (dimension)
5. delete it
6. FreeCAD crashes

If you edit the constraint everything goes fine but if you delete it, it crashes.

Thanks

discip, 2022-10-04:
This is not true for 2022.709.
Have you tried that?
Please consider closing this issue if it is solved. 😊

discip, 2022-10-04:
@pgilfernandez
This is not true for 2022.709.
Have you tried that?
Please consider closing this issue if it is solved. 😊

pgilfernandez, 2022-10-05:
I'm afraid it happens to me in that version and under macOS:
```
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)
```

discip, 2022-10-05:
@pgilfernandez
Sorry, I did not think about OSes other than win 10! 😔

pgilfernandez, 2023-02-13:
Hi again,

It got solved... I'm not sure when by recent builds doesn't crash anymore so let me close the issue.

Thank you!

closed 2023-02-13 by pgilfernandez