realthunder/FreeCAD #520: Transparent overlay widget background not transparent on Fedora with Adwaita?

scottmudge, 2022-07-29:
Built LinkDaily branch (as of today) on Fedora 36. Have the "vanilla" Adwaita Dark theme active for the OS.

The background of the Tree view and other overlayed/transparent widgets doesn't seem to be actually transparent. Which is weird, because the titlebars for them *are* transparent, as are the containers for the widgets.

Screenshot:

[image]

Anyone know a workaround for this? I've tried tweaking the stylesheets, but to no avail.

realthunder, 2022-07-29:
Can you please try `Tools -> Preset configurations -> Overlay dark theme`. If that does not work, please post your user settings here. `Tools -> Edit parameters -> Export`.

scottmudge, 2022-07-29:
Same result with the preset. I've attached my parameters below.

I think the issue is similar to https://github.com/realthunder/FreeCAD/issues/168 , but I can't determine if there was ever a solution. I suppose I could go through the rigamarole of installing a different desktop environment, but I rather like GNOME.

[params.zip](https://github.com/realthunder/FreeCAD/files/9223959/params.zip)

These same parameters look fine on my Windows build (same git node), so my assumption is there is some kind of compatibility issue?

scottmudge, 2022-07-30:
Ah, can confirm that using qt5ct and forcing the "Fusion" theme before launching FreeCAD does fix the transparency issue, but then the window decorations are no longer white and so don't match.

So it seems to be an issue with the Adwaita theme. I can try finding an alternative dark style which matches, but this seems to be the core of the issue.

scottmudge, 2022-07-30:
Alright, found a solution.

Setting the qt theme in qt5ct to Fusion, and then setting the "Palette" to custom with the "darker" palette gives a close enough approximation to the intended look with functional transparency.

If there is a programmatic way to detect Adwaita and enforce Fusion w/ the appropriate palette, that might be a way to improve compatibility for systems with Adwaita by default, but probably a low priority.

closed 2022-07-30 by scottmudge

scottmudge, 2022-07-30:
One final update, I managed to get transparency working with any arbitrary GTK/GNOME theme by installing qt5-style-plugins, and then setting the `QT_QPA_PLATFORMTHEME=gtk2` environmental variable.

qt5ct is no longer required, and it applies to any qt app.