realthunder/FreeCAD #524: Black dots on part when in shadow view

Jakeo-93, 2022-08-03:
[image]

os is arch linux and im using the snap pack. I have noticed after the 07/25 update that in shadow view depending on how the light hits the part these black dots are present now on the part.

The other issue ive noticed is when opening the preference box in edit there is a considerable amount of lag from the time you click it and the time it opens on the screen

ForgeThrall, 2022-09-30:
Issue replicated on Windows 10, using [Midnight UI](https://forum.freecadweb.org/viewtopic.php?t=55779) on a fresh install

realthunder, 2022-09-30:
The bug has been fixed in 7ac4493ad224d2ef670daee3ce558080defd608a.

Until the next release, as a workaround, after you've activated shadow mode, manually changed the Document property Epsilon to 0.00001

[image]

discip, 2022-10-05:
@Jakeo-93
Please remember closing solved issues. 😊

closed 2022-10-05 by Jakeo-93