realthunder/FreeCAD #526: Making a groove from a sketch clone into a feature-based body not (always) working anymore

not-a-number-dot-org, 2022-08-06:
The attached example is a subset of my project (Ariel loudspeaker) that works fine with the FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220430 version but doesn't work with the FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220709 version I just installed.

Just load, mark for refresh and refresh it.

I have a panel in which I make six grooves that are then partially filled in. I then have to make grooves that are positioned both on the panel and on the parts that have been filled in.
To do this I create a cover that serves as a base for six bodies on which I apply a clone of the groove at different vertical offsets.

This method works for five of the six covers... the first one whose assembly is slightly different from the others because it contains the body that serves as the base for all the covers.

[20220430.FCStd.gz](https://github.com/realthunder/FreeCAD/files/9275498/20220430.FCStd.gz)
.

realthunder, 2022-08-07:
There is a bug in previous version. The `SubShapeBind` in latest version is behaving as expected. Looks like the way you are using the `SubShapeBinder` requires `Relative` to be turned off. So before triggering recompute, change all the binder `Relative` to false first. When `Relative` if true, the shape binder will adjust the bound shape with the binder's parent and bound object's `Relative` placement. But your binder `Support` has no parent setup.

not-a-number-dot-org, 2022-08-07:
Indeed it worked.
Thanks!

discip, 2022-10-05:
@not-a-number-dot-org
According to your reaction, this issue should be solved.
Please remember to close solved issues, to maintain order. 😊

closed 2022-10-08 by not-a-number-dot-org