realthunder/FreeCAD #533: Sheet Metal WB not unfolding correctly in LinkStage3

thesonnenberg, 2022-08-15:
when trying to unwrap a sheet metal object the following error occurs and no sketch with bend lines is generated  
_Traceback (most recent call last):
  File "/home/xxxxx/snap/freecad-realthunder/common/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 2899, in accept
    grp1 = TechDraw.projectEx(s,norm)
<class 'AttributeError'>: module 'TechDraw' has no attribute 'projectEx'_

I could not recreate the same error with the FreeCAD 0.20 main branch, that is why my assumption is that it is not necessarily a Sheet Metal WB issue but might be related to the LinkStage3 branch.

attached the screenshots from the LinkStage3 branch when the error occurred.

**System/SW used:**
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.801.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.08.01-edge)
Hash: 1a5c42ee6bd3e32e3dfee8f659349546d217569a
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version:
 7.6.2
Locale: English/United States (en_US)

[image]
[image]

stuartpittaway, 2022-09-07:
I've also got the same issue on Windows using the version FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220709

luzpaz, 2022-10-07:
@realthunder is this asm3 or link related ?