realthunder/FreeCAD #535: Blend Surface is also not working properly (Note: In the latest version)

hamza00789, 2022-08-17:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US) 

Blend Surface is also not working properly in the new version. I updated the curve workbench and now the blend surface is not working properly either. Degradation is happening instead of progress. Pre-existing commands are not working properly. 

Old result

[image] 

New result 

[image]
 
File 
 
[1.zip](https://github.com/realthunder/FreeCAD/files/9359236/1.zip)

realthunder, 2022-08-18:
Can you please post which version is correctly working? Did that version working correctly after you updated Curve workbench?

hamza00789, 2022-08-18:
This is the old version in which the Blend surface works just fine. 

[Curves.zip](https://github.com/realthunder/FreeCAD/files/9373589/Curves.zip)    

You can see the result is correct. 

[image]

hamza00789, 2022-08-18:
This is an updated version of the curve workbench.  

[Curves.zip](https://github.com/realthunder/FreeCAD/files/9373615/Curves.zip) 

The blend surface doesn't work properly in it. You can see that the result is bad.  

[image]