realthunder/FreeCAD #537: Can not compile LinkDaily branch

ceremcem, 2022-08-19:


closed 2022-08-19 by realthunder

ceremcem, 2022-08-19:
I still can not compile: 

```
[ 64%] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/Workbench.cpp.o
/home/aea/FreeCAD/src/Gui/OverlayWidgets.cpp: In member function ‘void Gui::OverlayManager::Private::dragDockWidget(QPoint, QWidget*, QPoint, QSize, bool)’:
/home/aea/FreeCAD/src/Gui/OverlayWidgets.cpp:3842:33: warning: ‘dstDockArea’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                         if (dockWidget == dock
                             ~~~~~~~~~~~~~~~~~~
                                 || !dockWidget->isVisible()
                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 || getMainWindow()->dockWidgetArea(dockWidget) != dstDockArea)
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 65%] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/WorkbenchFactory.cpp.o
```

Starting the compilation again gives the following error: 

```
[ 61%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/DlgPointsReadImp.cpp.o
/home/aea/FreeCAD/src/Mod/Part/App/TopoShapeEx.cpp: In member function ‘Part::TopoShape& Part::TopoShape::makEEvolve(const Part::TopoShape&, const Part::TopoShape&, Part::TopoShape::JoinType, bool, bool, bool, double, const char*)’:
/home/aea/FreeCAD/src/Mod/Part/App/TopoShapeEx.cpp:1652:16: error: no matching function for call to ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved(TopoDS_Shape&, TopoDS_Wire&, GeomAbs_JoinType&, bool, bool, bool, double&)’
             tol);
                ^
In file included from /home/aea/FreeCAD/src/Mod/Part/App/TopoShapeEx.cpp:181:
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:91:19: note: candidate: ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved(const TopoDS_Face&, const TopoDS_Wire&, GeomAbs_JoinType, Standard_Boolean, Standard_Boolean, Standard_Boolean, Standard_Real)’
   Standard_EXPORT BRepOffsetAPI_MakeEvolved(const TopoDS_Face& Spine, const TopoDS_Wire& Profil, const GeomAbs_JoinType Join = GeomAbs_Arc, const Standard_Boolean AxeProf = Standard_True, const Standard_Boolean Solid = Standard_False, const Standard_Boolean ProfOnSpine = Standard_False, const Standard_Real Tol = 0.0000001);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:91:19: note:   no known conversion for argument 1 from ‘TopoDS_Shape’ to ‘const TopoDS_Face&’
In file included from /home/aea/FreeCAD/src/Mod/Part/App/TopoShapeEx.cpp:181:
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:69:19: note: candidate: ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved(const TopoDS_Wire&, const TopoDS_Wire&, GeomAbs_JoinType, Standard_Boolean, Standard_Boolean, Standard_Boolean, Standard_Real)’
   Standard_EXPORT BRepOffsetAPI_MakeEvolved(const TopoDS_Wire& Spine, const TopoDS_Wire& Profil, const GeomAbs_JoinType Join = GeomAbs_Arc, const Standard_Boolean AxeProf = Standard_True, const Standard_Boolean Solid = Standard_False, const Standard_Boolean ProfOnSpine = Standard_False, const Standard_Real Tol = 0.0000001);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:69:19: note:   no known conversion for argument 1 from ‘TopoDS_Shape’ to ‘const TopoDS_Wire&’
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:67:19: note: candidate: ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved()’
   Standard_EXPORT BRepOffsetAPI_MakeEvolved();
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:67:19: note:   candidate expects 0 arguments, 7 provided
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:60:7: note: candidate: ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved(const BRepOffsetAPI_MakeEvolved&)’
 class BRepOffsetAPI_MakeEvolved  : public BRepBuilderAPI_MakeShape
       ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:60:7: note:   candidate expects 1 argument, 7 provided
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:60:7: note: candidate: ‘BRepOffsetAPI_MakeEvolved::BRepOffsetAPI_MakeEvolved(BRepOffsetAPI_MakeEvolved&&)’
/usr/include/opencascade/BRepOffsetAPI_MakeEvolved.hxx:60:7: note:   candidate expects 1 argument, 7 provided
[ 61%] Built target SpreadsheetGui_autogen
[ 61%] Generating qrc_Image.cpp
Scanning dependencies of target ImageGui
[ 61%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/qrc_Image.cpp.o
[ 61%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/AppImageGui.cpp.o
[ 61%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/qrc_Points.cpp.o
make[2]: *** [src/Mod/Part/App/CMakeFiles/Part.dir/build.make:2459: src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapeEx.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2385: src/Mod/Part/App/CMakeFiles/Part.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 61%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/AppPointsGui.cpp.o
[ 61%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/Command.cpp.o
[ 61%] Building CXX object src/Mod/Test/Gui/CMakeFiles/QtUnitGui.dir/UnitTestPy.cpp.o
[ 61%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/AppImageGuiPy.cpp.o
[ 61%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/Command.cpp.o
[ 61%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/PreCompiled.cpp.o
[... cut ...]
[ 63%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/PreCompiled.cpp.o
[ 63%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/Workbench.cpp.o
[ 63%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/OpenGLImageBox.cpp.o
[ 63%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/ImageGui_autogen/mocs_compilation.cpp.o
[ 63%] Linking CXX shared library ../../../../Mod/Image/ImageGui.so
[ 63%] Built target ImageGui
make: *** [Makefile:130: all] Error 2
```

Commit: ff5ad0aefe (HEAD -> LinkDaily) Draft: fix Wire end points update

realthunder, 2022-08-20:
Please try again with the latest commit. I tried OCCT 7.6 and 7.5. You're probably using older version.

ceremcem, 2022-08-20:
Now I can succesfully compile. 

My configuration is as follows: 

```
-------------------------------
Building in /home/aea/fc-build/Release
-------------------------------

Calculated build command:
+ cd /home/aea/fc-build/Release
+ cmake ../../FreeCAD '-DFREECAD_USE_OCC_VARIANT=Official Version' -DCMAKE_BUILD_TYPE=Release -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_ENABLE_CXX_STD:STRING=C++17 -DBUILD_FEM_NETGEN=1
-- Compiler: GNU, version: 8.3.0
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.13.4
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
=====================================================
Doxygen not found, will not build documentation.     
=====================================================

-- PyCXX found:
--   Headers:  /home/aea/FreeCAD/src
--   Sources:  /home/aea/FreeCAD/src/CXX
--   Version:  6.2.8
-- -- Found OCE/OpenCASCADE version: 7.3.0
-- -- OCE/OpenCASCADE include directory: /usr/include/opencascade
-- -- OCE/OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- VTK components: vtkCommonCore;vtkCommonDataModel;vtkFiltersVerdict;vtkIOXML;vtkFiltersCore;vtkFiltersGeneral;vtkIOLegacy;vtkFiltersExtraction;vtkFiltersSources;vtkFiltersGeometry;vtkIOMPIParallel;vtkParallelMPI;vtkhdf5;vtkFiltersParallelDIY2;vtkRenderingCore;vtkInteractionStyle;vtkRenderingFreeType;vtkRenderingOpenGL2
-- Check for medfile (libmed and libmedc) ...
[... 76 lines cut ...]
-- Qt5Widgets:          5.11.3
-- Qt5PrintSupport:     5.11.3
-- Qt5OpenGL:           5.11.3
-- Qt5Svg:              5.11.3
-- Qt5UiTools:          5.11.3
-- Qt5Concurrent:       5.11.3
-- Qt5WebKitWidgets:    5.212.0
-- Shiboken2:           5.11.2 [/usr/lib/x86_64-linux-gnu/cmake/Shiboken2-5.11.2] [/usr/include/shiboken2]
-- PySide2:             5.11.2 [/usr/include/PySide2]
-- PySide2Tools:        [/usr/bin/pyside2-uic] [/usr/bin/pyside2-rcc]
-- Freetype:            2.9.1
-- OpenGL:              /usr/lib/x86_64-linux-gnu/libGL.so
-- OpenGLU:             [/usr/lib/x86_64-linux-gnu/libGLU.so][/usr/include]
-- Coin3D:              4.0.0 [/usr/lib/x86_64-linux-gnu/libCoin.so] [/usr/include]
-- Pivy:                0.6.5a0
-- SPNAV:               [/usr/lib/libspnav.so] [/usr/include]
-- Matplotlib:          3.0.2
-- Rift:                not enabled (BUILD_VR)
-- Doxygen:             not found
=================================================
Now run 'cmake --build /home/aea/fc-build/Release' to build FreeCAD
=================================================
```

Should I update the OCCT regardless of compilation status?

commit ad6fe7ba (2022-08-19): Part: fix compilation

Fixes #537
@ceremcem