realthunder/FreeCAD #541: Refine doesn't work in some places

hamza00789, 2022-08-21:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
 
Refine is not working in Extrude either. These edges should disappear in Extrude.               

[image]

hamza00789, 2022-08-21:
similar issues 
  
#336  #426 #538 #539

Miksu80, 2022-12-13:
I noticed this too, here the circles should be flat planes, the newest version has gone worse:
[image]

here is the design:
[Vilamenttistandi.zip](https://github.com/realthunder/FreeCAD/files/10218520/Vilamenttistandi.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: Finnish/Finland (fi_FI)