realthunder/FreeCAD #544: [Bug] Unclickable navigation cube

YosypC, 2022-08-23:
https://user-images.githubusercontent.com/50027752/186245445-3c8b2322-89be-486f-9ac2-2782bc3b1da7.mp4

Clicking on the navigation cube yields no action: no face or corner highlighting, no face or corner alignment, no menu shown, can't rotate the model.
It happens with new documents as well.

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)
```

realthunder, 2022-09-02:
That's strange. Did this always happen on your computer, or does it only happen in some condition?

YosypC, 2022-09-02:
Negative, it has always worked. Now that I think about it, it probably stopped working when I installed and opened the FC Master Branch. (this one:

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/Europe (en_150)
Installed mods: 
  * BOLTSFC
  * ExtMan
  * fasteners
  * fcgear
[/code]
)

I've checked and it doesn't work there neither.

closed 2026-01-13 by YosypC