realthunder/FreeCAD #548: Cyclic dependency / still touched after recompute / the graph must be a dag after moving a sketch in Tree View

YosypC, 2022-08-31:
[Case.zip](https://github.com/realthunder/FreeCAD/files/9460738/Case.zip)

The problem is in **PCI-MBCutout** body, in _Case_ -> _MainSection_. Every time I delete something, the features automatically go out of their body, and I get the errors in the title even when I try to bring them back in. I moved **Sketch14** to the end since I needed the edges of **Pad012**. I'm stuck with a broken model.

`OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)
`

realthunder, 2022-09-03:
I have improved cyclic dependency diagnostic output. For your attached file, open it and recompute it and see the detailed output in the report view, as shown below,

```
Dependency cycles:
MiniITXCase#Body008 (PCI-MBCutout), Property: Tip, Group[12]
MiniITXCase#Pad012, Property: UpToFace
MiniITXCase#Pad010, Property: Profile
MiniITXCase#Sketch012, Property: ExternalGeometry[0]
MiniITXCase#Import005 (Import005(Pad009)), Property: Support
```

The cycle is formed as below 
* `Body008.Tip (also Body008.Group[12]) -> Pad012`
* `Pad012.UpToFace -> Pad010`
* `Pad010.Profile -> Sketch012`
* `Sketch012.ExternalGeometry[0] -> Import005`
* `Import005.Support -> Body008`

YosypC, 2022-10-05:
There's a lot of doubts I have about cyclic dependency, the reason why the features go out of the body and even the report view (I have never been able to see detailed output), but my main issue was solved. I guess I'll close the case.

closed 2022-10-05 by YosypC