realthunder/FreeCAD #552: Assembly 3 Error Windows 10

ThanosHusk, 2022-09-08:
Greetings, 
I am trying to assemble some part using assembly 3 and as soon as I place the first constraint the following error appears. The same error happens with all the constraints. 

16:03:51  Traceback (most recent call last):
  File "C:\Users\thano\OneDrive\Desktop\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220709\Mod\asm3\freecad\asm3\assembly.py", line 333, in claimChildren
    return getattr(self.ViewObject.Object, 'Group', [])
<class 'AttributeError'>: 'ViewProviderAsmConstraint' object has no attribute 'ViewObject'

I am not new to CAD in general, however I am fairly new to FreeCAD. I also asked a person close to me that knows FreeCAD and he doesn't seem to find the cause of this error. 
Any help would be appreciated!

realthunder, 2022-09-09:
This is a harmless error. You can ignore it for now. I'll fix it in the next release.

ThanosHusk, 2022-09-09:
That's great! However the about-to-be-assembled components freeze in place and the chosen surfaces become red. Maybe there's another error I don't quite see there and it doesn't have to do with the one I mentioned above?

realthunder, 2022-09-09:
The 'ViewObject' error only occurs when creating a new empty assembly. Further action the assembly looks fine. Can you please post a screen cast showing the problem you have?

ThanosHusk, 2022-09-09:
The same error appears each time i place a new constraint also. 

[image]

realthunder, 2022-09-09:
I see. It is indeed a problem. I'll make a new release soon.

chetanpm, 2023-02-14:
Unable to assemble at all.
OS: macOS 11.6
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)

[image]