realthunder/FreeCAD #557: Linear pattern - cannot select reference

jj-uk, 2022-09-18:
To reproduce:

Part design:
Create a body - create a sketch - XY plane.
Create a rectangle, close sketch and pad.

Select the end of the rectangle pad, sketch on face.
[image]

Bring in geometry, Add a rectangle to the side, pad.
[image]
[image]

Select pad, linear pattern, select reference, select edge or 1st pad. Nothing happens. 
[image]

This works ok in freecad 0.20, but not in daily or stable.

discip, 2022-10-05:
@realthunder
That is absolutely true!

This prevented me from getting my job done today!

Thanks in advance

hobbymachinist, 2024-07-23:
@realthunder ran into the same issue today, is there any workaround for this ? seems to work fine in Freecad 0.21.2

jbayless, 2024-07-24:
@hobbymachinist I have written a very, very bodged workaround here on this duplicate of the issue: https://github.com/realthunder/FreeCAD/issues/710

hobbymachinist, 2024-07-25:
Yikes, that's a gnarly workaround. I'm going to switch back to using Freecad 0.21.2 for now.

closed 2024-08-28 by realthunder

discip, 2024-08-28:
@realthunder
# Great, thank you very much! 😃👍

commit b50ff630 (2024-08-28): Gui: fix recursive selection notification

Fixes #557
Fixes #710
commit a6d27c43 (2024-08-28): Gui: fix recursive selection notification

Fixes #557
Fixes #710