realthunder/FreeCAD #561: HiDPI toolbar icons are twice too big

m-sundman, 2022-09-22:
On normal FreeCAD 0.20.1 if I select 16px toolbar icons they end up being about 32px. On rt's FreeCAD 2022.07.09 if I select 16px toolbar icons they end up 64px.
I'm running KDE (Kubuntu 22.04).
`xrandr` shows correctly that _"2560x1600+0+0 (normal left inverted right x axis y axis) 345mm x 215mm"_ which is the correct resolution, and more or less exactly the right size of this 189 dpi screen.
I'm also using a global scale of 150% in the KDE display settings.
(xdpyinfo doesn't know what's what, and thinks I have 96 dpi, but this seems to be a common problem, and is probably completely unrelated)
`xrdb -query` shows 144 dpi (which is 150% of 96), so semi-clueless.

m-sundman, 2022-09-22:
OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

m-sundman, 2022-09-22:
Setting `<FCBool Name="RespectSystemDPI" Value="0"/>` to `<FCBool Name="RespectSystemDPI" Value="1"/>` in `link.user.cfg` makes no difference.

m-sundman, 2022-09-22:
Other UI elements, like this one, are also too big:
[image]

xplwowi, 2022-09-22:
The same OS + DM, also 150% global scale, DPI forced to 152, 4k screen 3840x2160.
The problem only occurs when running FC from the appimage package. I'm using the self-compiled version and it's okay there.
[image]

xplwowi, 2022-09-22:
BTW, there are many applications that enforce some combinations when scaling incorrectly. Fortunately, QT is here so you can try to force environment variables. Appimage FC displays correctly for me when I invoke it like this: 
QT_SCREEN_SCALE_FACTORS=1 QT_FONT_DPI=144 ./FreeCAD

DraftSight forced me to find experimentally such (contrary to logic) environment variables combination:
PLASMA_USE_QT_SCALING=0 QT_AUTO_SCREEN_SCALE_FACTOR=0 QT_SCREEN_SCALE_FACTORS=0.5 QT_SCALE_FACTOR=4 QT_FONT_DPI=72 ./DraftSight

m-sundman, 2022-09-22:
> BTW, there are many applications that enforce some combinations when scaling incorrectly. Fortunately, QT is here so you can try to force environment variables. Appimage FC displays correctly for me when I invoke it like this: QT_SCREEN_SCALE_FACTORS=1 QT_FONT_DPI=144 ./FreeCAD

Ah, very nice! Since I use a general screen scaling of 150% and my real DPI is 189 then I can use a QT_SCREEN_SCALE_FACTORS of 1 and a QT_FONT_DPI of 126 (which is 189/1.5), and this seems to work pretty well with the appimage. Thanks!

discip, 2022-10-05:
@m-sundman
Is this issue solved?
If so, please consider closing it. 😊

m-sundman, 2022-10-06:
Well, the "solution" above is more of a workaround, which has to be applied manually to each version of the appimage I install (until the problem is fixed for real). The actual problem is still there.