realthunder/FreeCAD #562: Multiple issues with fit tolerance and TopoShape

jccates, 2022-09-25:
I'm working on a design for a chair to be milled on a CNC router, and it uses a lot of tab-and-slot joints to join non-coplanar parts together. Most of the slots are defined as pockets generated from sketches attached to the surface of the base feature, like this:
[image]
My prototype had fit issues, so I've started using the "Fit tolerance" parameter in the Pocket dialogue to make it easier to put parts together. Since the 9/23/2022 release I've been seeing errors whenever I try to recompute parts with fit tolerances. There are two types of errors; the first comes from TopoShape::MakEShell, and I see it in instances like the image I posted, where there is more than one closed wire in the sketch, so we end up with more than one pocket shape:

`19:47:52  2537.01 <App> Document.cpp(4327): Failed to recompute spike_chair#Pocket003: Failed to make shell: unexpected output shape type 8`

I'm guessing MakEShell is getting a shape with type TopAbs_COMPOUND or something like that, so we descend into the conditional that produces this error.

The second type of error (this time from TopoShape::getSubTopoShape), happens if I edit the sketch to produce only one closed wire:

`19:11:22  346.777 <App> Document.cpp(4327): Failed to recompute spike_chair#Pocket051: Shape index 1 out of bound 0`

Fit tolerance expands the pocket shape(s); I'm guessing the bounds are not being recomputed after tolerance is applied and that's why we're hitting this. I see the same error if I try to apply tolerance to a Pad.

realthunder, 2022-09-25:
Can you please post your FreeCAD About information here? Also, is it possible to share your file?

jccates, 2022-09-25:
Here's the About panel:
[image]
The original is too large for github, but I was able to make a smaller version by removing the Path jobs from the project, which aren't relevant to this issue anyways:
[spike_chair_nosheet.FCStd.gz](https://github.com/realthunder/FreeCAD/files/9639929/spike_chair_nosheet.FCStd.gz)

closed 2022-09-25 by realthunder

jccates, 2022-09-29:
works great since the fix; thanks for the quick response!

commit f924a50a (2022-09-25): Part: fix TopoShape::makERefine() with slient fail

Fixes #562
@jccates