realthunder/FreeCAD #563: Recomputaion of existing project is failed

vovodroid, 2022-09-25:
I'm opening document created in FreeCad 0.20.1 and recomputation failes with error 
`09:52:57  0 <App> Document.cpp(4262): Failed to recompute Direct#AdditiveLoft: Loft: At least one section is needed`

It happens in both latest stable and development releases.

Project attached.

[Direct.zip](https://github.com/realthunder/FreeCAD/files/9640188/Direct.zip)

discip, 2022-10-05:
I noticed this as well.
For some reason the loft function 'forgets' one of the two needed sketches, if the file was not created with ***'LinkCAD'***.

The solution is not that hard. In your case you just need to reassign 4 sketches to the right loft.

[Direct.zip](https://github.com/realthunder/FreeCAD/files/9716837/Direct.zip)

discip, 2022-10-05:
@realthunder
I noticed that as well.
For some reason the loft function 'forgets' one of the two needed sketches, if the file was not created with ***'LinkCAD'***.

@vovodroid
The solution is not that hard. In your case you just need to reassign 4 sketches to the right loft.

Here is the fixed version:
[Direct.zip](https://github.com/realthunder/FreeCAD/files/9716837/Direct.zip)

discip, 2022-10-05:
@realthunder
I noticed that as well.
For some reason the loft function 'forgets' one of the two needed sketches, if the file was not created with ***'LinkCAD'***.

@vovodroid
The solution is not that hard. In your case, you only need to reassign 4 sketches to the respective loft.

Here is the fixed version:
[Direct.zip](https://github.com/realthunder/FreeCAD/files/9716837/Direct.zip)

closed 2024-04-25 by vovodroid