realthunder/FreeCAD #567: zooming using mouse scroll wheel

bambukouk, 2022-10-01:
I am using:
```
OS: Ubuntu Core 20 (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.926.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.09.26-edge)
Hash: c1616d234998bdceea714c5bdf21329342f05219
Python version: 3.8.10
Qt version: 5.15.5
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United Kingdom (en_GB)
```
Display preferences set to:
`3D Navigation = CAD`

**Zooming using mouse scroll wheel doesn't work when the cursor is away from window centre.** 

To narrow down the issue, tested using:
```
OS: Ubuntu Core 20 (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 0.21.30490 (Git) Snap 325
Build type: Release
Branch: master
Hash: 7669bd8adb82bcb0ed13af1f5426d259b6e838dc
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

and in this version everything is perfectly fine - zooming using mouse scroll wheel works with the cursor anywhere on the screen.

realthunder, 2022-10-02:
Can you please post a screencast showing the problem? Also please post your user settings here, `Tools -> Edit parameters -> Export`.

bambukouk, 2022-10-02:
My settings file (renamed .zip to permit attaching here):
[2022_10_02.zip](https://github.com/realthunder/FreeCAD/files/9692011/2022_10_02.zip)

I also made some progress in identifying the problem!
It's not the cursor in the centre ve cursor elsewhere...
see below:

[image]

[image]

closed 2022-10-03 by realthunder

bambukouk, 2022-10-03:
Thank you as always!
Your work is brilliant :+1:

commit a6643045 (2022-10-03): Gui: fix mouse wheel handling in overlay widget

Fixes #567
@bambukouk