realthunder/FreeCAD #569: Feature request: add 3D sketch rotation

vovodroid, 2022-10-02:
Original Freecad has free sketch rotating in 3D. It's really handy option, it would very good to merge this from original branch.

realthunder, 2022-10-03:
I am not sure what do you mean by 'free sketch rotating in 3D'. Can you please post a screen cast?

vovodroid, 2022-10-03:
While you are in sketcher you can rotate it together in all directions with the whole body as on any other screens:

[image]

realthunder, 2022-10-03:
Try holding 'Alt' key and then do the rotation?

vovodroid, 2022-10-03:
Indeed, thank! Actually I looked for navigation style without Shift-Alt modifiers. In original FreeCad I use Tinkercad style, and here I found Gesture style.

Could it be customized in preferences?

realthunder, 2022-10-03:
Unfortunately, no. The navigation style needs a lot rework for that to happen

discip, 2022-10-05:
@vovodroid
It seems that your initial problem has been solved.
Please don't forget to close it. 😊