realthunder/FreeCAD #570: changing style sheet causes very long response time for opening preference window.

Jakeo-93, 2022-10-02:
Arch linux. 
feecad 2022-09-26

Changing the style sheet in the last few releases has caused significant hang time when re opening the preferences box. close to over 1 minute if not longer
[image]

realthunder, 2022-10-03:
Can you please paste your About information here? And which stylesheet are you using? Can you clarify the hang time here? Are you saying that the stylesheet changing is okay, but it will hang when you close the preference dialog and reopen it?

Jakeo-93, 2022-10-05:
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.926.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.09.26-edge)
Hash: c1616d234998bdceea714c5bdf21329342f05219tyle s
Python version: 3.8.10
Qt version: 5.15.5
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/Canada (en_CA)

and yes what im saying is when i switch the syle sheet it works well... but after im done closing the style sheet and i attempt to reopen the preference box is when the hang time occurs. And this is using the style sheets that come with a fresh install. dark orange or any of the other ones to be exact