realthunder/FreeCAD #574: Some feature requests

discip, 2022-10-04:
@realthunder
Having used at least 4 different CAD programs, I have always liked one or the other feature in this program or the other.
And I was wondering if it would be possible to combine all the great features in one piece of software.

I really don't know if it is allowed to 'copy' functions from other software. So if this is not allowed please consider closing this FR.

Here is a small list of features that I think are worth implementing in FreeCAD if possible:

1. Show in **Sketcher** possible constraints _depending_ on which _line_, _vertex_ or _curve_ the mouse _hovered_.

2. Combine `Horizontal Distance`, `Vertical Distance`, `Arc`... into one _smart_ ***'Dimension Tool'*** that depending on your selections (e.g. vertex & line) and the angle to the respective selections proposes either a 'Horizontal Distance ', an 'Angle'...

3. Allow adding _references_ while typing **formulas** in by _clicking_ on the _corresponding value_ which is revealed in sketch style after the specific _sketch_ or function (e.g. pad) is selected.

Hope this makes sence! 😃

Thanks in advance
kind regards

luzpaz, 2022-10-07:
Please don't group feature requests in to one ticket. instead make a ticket for each one, @discip 

Closing

closed 2022-10-07 by luzpaz

discip, 2022-10-07:
@luzpaz
Done. 😊