realthunder/FreeCAD #580: Crash caused by selecting a face for the first transform (a mirror in this case) of a multi-transform

ardugitter, 2022-10-08:
I am trying to mirror a part in 2 directions, one after the other. In attempting the first mirror transform, FreeCAD crashes if you select this face as the one to mirror about:

[image]

This is the file in question:

[Corner protectors.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9734745/Corner.protectors.FCStd.zip)

Version info:

```
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

Thanks.

realthunder, 2022-10-08:
Firstly, the way you use the MultiTransform feature seems questionable. Are you trying to mirror the whole shape? If so, please uncheck the `Transform sub-featre` option.

This crash problem is most likely a bug in OCC. The Mirrored shape is invalid for some reason. The crash happens when try to fuse the invalid mirrored shape.

ardugitter, 2022-10-08:
Yes, I am trying to mirror the entire shape.

If I untick the subfeature box in the file I provided, it crashes..

If I delete the multi-transform and start again, creating another multi-transform with subfeature unticked, I can create 2 mirror transforms as I desire, but when I click "OK" it crashes.

realthunder, 2022-10-08:
Yes I can reproduce that, and the crash happens inside the CAD kernel. I'll dig a bit deeper later, but I doubt there is much I can do.

realthunder, 2022-10-08:
BTW, `MultiTransform` is kind of deprecated in my branch. You can simply create normal transform features one after another, and enable `Transform sub feature` to transform the previous transformed sub-feature.

ardugitter, 2022-10-09:
I tried the normal mirror operation too and that crashed as well. :-)

Thank you for having a look into it. Your work is much appreciated.