realthunder/FreeCAD #587: total crashes without warning

yahbluez, 2022-10-13:
OS: Ubuntu Core 20 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.1008.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.10.08-edge)
Hash: 39bee7cbb376bae80aee89f7acd20931708c96cd
Python version: 3.8.10
Qt version: 5.15.5
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

This version is crashing if you just delete something from the tree view.

Deleting using the undo function works without crash.
But just click and delete a object in tree view crashes the program. 

This behavior happens since a couple of version.
I guess it is something very small but this bug disturbs the workflow a lot.

realthunder, 2022-10-13:
Can you please launch FreeCAD in a terminal, and then paste the printed call stack information after crash?

yahbluez, 2022-10-14:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f1da31ed090]
#1  /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so(+0x9aff35) [0x7f1da6282f35]
#2  0x7f1da6284598 in Gui::DAG::Model::selectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x138
#3  0x7f1da6617142 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x32
#4  0x7f1da627d80f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x31f
#5  0x7f1da6618b86 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x76
#6  0x7f1da627d80f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x31f
#7  0x7f1da6619ffc in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x4ec
#8  0x7f1da661a6d0 in Gui::SelectionSingleton::flushNotifications() from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x1d0
#9  0x7f1da661a7ea in Gui::SelectionPauseNotification::~SelectionPauseNotification() from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x2a
#10  0x7f1da60369b4 in StdCmdDelete::activated(int) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0xb94
#11  0x7f1da6025f11 in Gui::Command::_invoke(int, bool) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x241
#12  0x7f1da6026395 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x135
#13  /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ebd50) [0x7f1da4052d50]
#14  0x7f1da4b9f9a6 in QAction::triggered(bool) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#15  0x7f1da4ba2078 in QAction::activate(QAction::ActionEvent) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa8
#16  0x7f1da608511f in Gui::ShortcutManager::onTimer() from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0xdf
#17  0x7f1da6085d59 in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x599
#18  0x7f1da6086b16 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x246
#19  0x7f1da401b89e in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x8e
#20  0x7f1da4ba5ec0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc0
#21  0x7f1da4baec98 in QApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#22  0x7f1da5fb9df8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x88
#23  0x7f1da401bc6a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#24  0x7f1da446e323 in QShortcutMap::dispatchEvent(QKeyEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x463
#25  0x7f1da446ecc3 in QShortcutMap::tryShortcut(QKeyEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x63
#26  0x7f1da441ab32 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x192
#27  0x7f1da443bad1 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x91
#28  0x7f1da4440d8b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x24b
#29  0x7f1da441766c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#30  /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x783ee) [0x7f1d98ec63ee]
#31  /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f1da110517d]
#32  /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7f1da1105400]
[... cut ...]
#35  0x7f1da401a76b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#36  0x7f1da4022914 in QCoreApplication::exec() from /snap/freecad-realthunder/79/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#37  0x7f1da5f06dd9 in Gui::Application::runApplication() from /snap/freecad-realthunder/79/usr/lib/libFreeCADGui.so+0x3e9
#38  /snap/freecad-realthunder/79/usr/bin/FreeCADLink(+0x4ed0) [0x55f5d7f89ed0]
#39  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f1da31ce083]
#40  /snap/freecad-realthunder/79/usr/bin/FreeCADLink(+0x52ce) [0x55f5d7f8a2ce]

yahbluez, 2022-10-14:
What i did was:

1. start fc
2. new doc
3. new part design body
4. new sketch
5. close sketch
6. new body
7. new sketch
8. draw something in the sketch
9. close sketch
10. delte this (second) body
11. crash

realthunder, 2022-10-15:
The above commit shall fix the crash, which happens in `DAGView`. BTW, are you actively using `DAGView`? This view is not actively maintained right now, and is quite complex. To avoid unexpected behavior, you might want to disable it if not using. You can do this by using `Tool -> Edit parameters`. And set parameter `BaseApp/Preferences/DockWindows/DAGView/Enable` to `False`.

closed 2022-10-15 by realthunder

yahbluez, 2022-10-15:
I used it one time, but did not need it regularly. So i can switch it off with no harm of workflow. Thx for fixing.

commit 410ded63 (2022-10-15): Gui: fix crash in DAGView when deleting object

Fixes #587