realthunder/FreeCAD #589: Crash after deleting a boolean augument

wsteffe, 2022-10-15:
Crash occurs pressing delete of selected object shown in following image:
[image]

Output of bash terminal:

Assembly4 workbench loaded 
A3 available
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f45ae201520]
#1  0x7f45b366ba42 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, ElementEntry>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, ElementEntry> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::size() const from /usr/local/lib/libFreeCADGui.so+0x10
#2  0x7f45b3669288 in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, ElementEntry, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, ElementEntry> > >::size() const from /usr/local/lib/libFreeCADGui.so+0x1c
#3  0x7f45b3663709 in SoFCRenderCacheManager::removeSelection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) from /usr/local/lib/libFreeCADGui.so+0x7e5
#4  0x7f45b36bcec8 in Gui::SoFCUnifiedSelection::Private::checkSelection(Gui::SelectionChanges::MsgType, App::SubObjectT const&) from /usr/local/lib/libFreeCADGui.so+0xc6
#5  0x7f45b36bde41 in Gui::SoFCUnifiedSelection::Private::doAction(SoAction*) from /usr/local/lib/libFreeCADGui.so+0x72f
#6  0x7f45b36bca0f in Gui::SoFCUnifiedSelection::doAction(SoAction*) from /usr/local/lib/libFreeCADGui.so+0x6d
#7  0x7f45b36e99de in Gui::SoFCSelectionAction::callDoAction(SoAction*, SoNode*) from /usr/local/lib/libFreeCADGui.so+0x34
#8  0x7f45adbed095 in SoAction::traverse(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#9  0x7f45b36e99a7 in Gui::SoFCSelectionAction::beginTraversal(SoNode*) from /usr/local/lib/libFreeCADGui.so+0x27
#10  0x7f45adbed6e8 in SoAction::apply(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xc8
#11  0x7f45b38190f8 in Gui::View3DInventorViewer::onSelectionChanged(Gui::SelectionChanges const&) from /usr/local/lib/libFreeCADGui.so+0x45c
#12  0x7f45b3ae81a3 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /usr/local/lib/libFreeCADGui.so+0x67
#13  0x7f45b3b13912 in boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>::operator()(Gui::SelectionObserver*, Gui::SelectionChanges const&) const from /usr/local/lib/libFreeCADGui.so+0x74
#14  0x7f45b3b12a1f in void boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::rrlist1<Gui::SelectionChanges const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>&, boost::_bi::rrlist1<Gui::SelectionChanges const&>&, int) from /usr/local/lib/libFreeCADGui.so+0x6f
#15  0x7f45b3b119e5 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> > >::operator()<Gui::SelectionChanges const&>(Gui::SelectionChanges const&) from /usr/local/lib/libFreeCADGui.so+0x53
#16  0x7f45b3b108d0 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> > >, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) from /usr/local/lib/libFreeCADGui.so+0x2f
#17  0x7f45b356520a in boost::function1<void, Gui::SelectionChanges const&>::operator()(Gui::SelectionChanges const&) const from /usr/local/lib/libFreeCADGui.so+0x70
#18  0x7f45b3562dc4 in boost::signals2::detail::void_type boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void (Gui::SelectionChanges const&)>, 0u, Gui::SelectionChanges const&>(boost::function<void (Gui::SelectionChanges const&)>&, boost::signals2::detail::unsigned_meta_array<0u>, std::tuple<Gui::SelectionChanges const&> const&, boost::enable_if<boost::is_void<boost::function<void (Gui::SelectionChanges const&)>::result_type>, void>::type*) const from /usr/local/lib/libFreeCADGui.so+0x3a
#19  0x7f45b355fc7e in boost::signals2::detail::void_type boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void (Gui::SelectionChanges const&)>, Gui::SelectionChanges const&, 1ul>(boost::function<void (Gui::SelectionChanges const&)>&, std::tuple<Gui::SelectionChanges const&> const&, mpl_::size_t<1ul>) const from /usr/local/lib/libFreeCADGui.so+0x32
#20  0x7f45b355be8d in boost::signals2::detail::void_type boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >(boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> > const&) const from /usr/local/lib/libFreeCADGui.so+0x5b
#21  0x7f45b355741b in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> >::dereference() const from /usr/local/lib/libFreeCADGui.so+0x67
#22  0x7f45b3552468 in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> >::reference boost::iterators::iterator_core_access::dereference<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, [... cut ...]
#96  0x7f45af36475b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#97  0x7f45af36ccf4 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#98  0x7f45b2fe22d1 in Gui::Application::runApplication() from /usr/local/lib/libFreeCADGui.so+0x62b
#99  FreeCAD(+0xaa84) [0x561c42870a84]
#100  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f45ae1e8d90]
#101  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f45ae1e8e40]
#102  FreeCAD(+0x9aa5) [0x561c4286faa5]

closed 2022-10-15 by realthunder

wsteffe, 2022-10-15:
Here annexed also the related FC document
[CoaxFilt_7_2_v7.zip](https://github.com/realthunder/FreeCAD/files/9791757/CoaxFilt_7_2_v7.zip)

I have just reverted back to  commit 16cf7b0171fab8b2c6c89ddda0678759eac6d8d8 which has fixed this problem.

realthunder, 2022-10-15:
Please try with the above commit and see if it is fixed.

wsteffe, 2022-10-15:
Yes, sorry. I had written last post without seeing you had already fixed.

commit 7a7961b4 (2022-10-15): Renderer: fix crash on remove selection

Fixes #589
@wsteffe