realthunder/FreeCAD #594: invalid subname. What does it mean ?

wsteffe, 2022-10-18:
[image]

I get this error every time I am updating a spreadsheet containing a few design parameters. I seems anyway that the geometry is properly updated.

realthunder, 2022-10-18:
It is from some external workbench, or Macro. Can you please search your `.FreeCAD` directory for this FCvarUpdate.py file and let me know which workbench/Macro it belongs.

realthunder, 2022-10-18:
Is it possible to share your file?

wsteffe, 2022-10-18:
yes sure:
[CoaxFilt_7_2_doubleSymmetry.zip](https://github.com/realthunder/FreeCAD/files/9811453/CoaxFilt_7_2_doubleSymmetry.zip)

realthunder, 2022-10-19:
I can't seem to reproduce the problem. I used FCvarUpdate for updating the parameters but no warning message. Can you please describe your workflow. Maybe also post your settings here `Tools -> Edit prarameters -> Export`.

BTW, make sure you build is synced to my branch.

wsteffe, 2022-10-19:
Hi RT, here are my parameters:
[myparam.zip](https://github.com/realthunder/FreeCAD/files/9819357/myparam.zip)
Looking at those parameters you may also perhaps tell me why I can't see the mapped name (but just a simple name such as Face1) when I select a 3d object. 

My build was synced with 0a9e2bbf71086f54d67319b0879ba73bd1c9380b. But I have just started to build with last commit (a3d793869825443b46d981707dbdbe8b8d72b168).

wsteffe, 2022-10-19:
Sorry RT. I have just seen that I sent the wrong FC file.
This was the file producing the error message [CoaxFilt_7_2_v7.zip](https://github.com/realthunder/FreeCAD/files/9819493/CoaxFilt_7_2_v7.zip)

wsteffe, 2022-10-19:
To complete the data I am annexing also the unmodified csv file:
[CoaxFilt_7_2_v7_var.csv](https://github.com/realthunder/FreeCAD/files/9819505/CoaxFilt_7_2_v7_var.csv)

To reproduce slightly change a variable and then execute FCvarUpdate.py.

commit dee714ea (2022-10-20): App: prevent warning message in GeoFeature::resolveElement()

Fixes #594
@wsteffe
closed 2022-10-20 by realthunder