realthunder/FreeCAD #595: Decreasing pad length causes previous feature disappear

vovodroid, 2022-10-19:
Decrease pad length a bit:

[image]

See what happens:

[image]

[DirectBack.zip](https://github.com/realthunder/FreeCAD/files/9818126/DirectBack.zip)

realthunder, 2022-10-25:
You model is pretty broken. It takes some effort to fix it.

The root cause of the problem is in `Sketch001`. For some reason, you created some empty edge, Edge11, Edge13 (i.e. the end vertices coincide), which causes invalid faces generated when extruding. You can see the problem by using `Part -> Check geometry` command on `Pad001`. The problem got carried over for multiple steps, and finally cause visible defects when as you've shown here.

Now, it's kind of tricky to fix your problem. You can simply delete `Edge11, Edge13` to fix `Sketch001` and `Pad001`. But as a result, the face that attaches `Sketch001` got rotated (actually, the internal plane geometry representation of the face is changed without visual differences), resulting the rotation of `Sketch001` as well. By right, the solver should relocate all the geometry accordingly, but it fails to do so with such a big jump, and misleadingly reporting redundant constraints). This is expected, as the solver is usually designed to handle gradual incremental changes, not huge jumps. You can manually rotate `Sketch001.AttachmentOffset` to fix the problem. The same things happens for `Sketch015`.

vovodroid, 2022-10-25:
> You model is pretty broken

Completely agree ))) But it would be nice to get error/warning on early stages.

By a way, what is current retaliation between your fork and original code? Your version has some great features, but where it's kept updated with latest FreeCAD?

closed 2024-04-25 by vovodroid