realthunder/FreeCAD #596: Again on linear history and DAG.

wsteffe, 2022-10-20:
Quite often I am struggling trying to restructure/change an existing design.

Typically I start deleting some features in a body because I want replace them with different features. This operation may break many other features also in other bodies which depend (through references) on the deleted objects. Initially I do not care about that because I have the intention to readjust the broken reference after having fully adjusted/restructured the first body.
But sometime I find myself into a messy situation in which FC give the error message "The graph must be a DAG" and  **I am not able to understand anymore the dependency structure and where I have to start to untangle the yarn**.

The here annexed file is an example of this mess. Please look only at first body named RODS.

Why, opening of sketch018 used to define Pad "Reson_5_6". I get the message
"The graph must be a DAG" ? This sketch refers only to two circles defined in the first sketch (named Sketch").

And why the structure generated by pad "Reson_5_6"includes also the two rods which were supposed to be generated by Reson_7/Sketch019 just following in the structure tree ?

I know that the tree order doesn't reflect the history and that the DAG doesn't need to have a linear structure. **But there should be a way to enforce that the Pad named "Reson_5_6" is added to the result of Pad named "Reson_3_4" following the tree order.**  (which doesn't seem to be in the generated structure).
 

 
[CoaxFilt_7_2_bad.zip](https://github.com/realthunder/FreeCAD/files/9829143/CoaxFilt_7_2_bad.zip)

realthunder, 2022-10-20:
The report view has detailed information on the cyclic dependency. In your case, it is show as follow,

```
Dependency cycles:                                                                                                
CoaxFilt_7_2_bad#Pad025 (Reson_5_6), Property: BaseFeature                          
CoaxFilt_7_2_bad#Pad026 (Reson_7), Property: BaseFeature   
```
What it means is `Reson_5_6` has a property `BaseFeature` that points to the feature in next line, i.e. `Reson_7`, which has a property `BaseFeature` points to the next  feature, and since it is the last line, so it is pointing to the first feature in the cycle, i.e. `Reson_5_6`. In case you can't find the property. It may be hidden. Either type code in Python console, or, right click the property view and choose `Show all`.

Can you recall what operation sequence you did to result in the cycle?

wsteffe, 2022-10-20:
I have checked Show all and that's what I see:

[image]
[image]

There isn't any BaseFeature

wsteffe, 2022-10-20:
Sorry, I should have looked under Data tab

wsteffe, 2022-10-20:
And it is also editable as I was asking for.
[image]

wsteffe, 2022-10-20:
Thanks anyway for help

wsteffe, 2022-10-20:
I can not recall how I have generated the dependency loop. But I have done some reordering of feature in the body and I have deleted some features in the same body.
Reordering is always a little bit problematic. Especially when I am trying to reorder a sketch. It always gives me error messages even if the sketch is actually moved to a different position (not always where I want to be)