realthunder/FreeCAD #598: Wrong external geometry link

vovodroid, 2022-10-21:
Though for BackSupportPad (Sketch007) external edge of PlatePad (Pad) is chosen it's linked to BackPad (Pad0001)
[image]

May be it causes #595 
[DirectBack.zip](https://github.com/realthunder/FreeCAD/files/9841147/DirectBack.zip)

realthunder, 2022-10-25:
How did you select the reference? You are supposed to manually unhide `Pad` and then select the edge. FreeCAD feature contains all the geometry from previous features.

vovodroid, 2022-10-25:
> How did you select the reference?

I don't remember. Just sow that I can't move Pad0001 because it has dependency, and found, that external edge linked to wrong object (though visually it's in right place).

closed 2024-04-25 by vovodroid