realthunder/FreeCAD #599: 2022.10.21 releases misses stable AppImage

gilbsgilbs, 2022-10-22:
Hi,

Unlike previous releases, the [2022.10.21](https://github.com/realthunder/FreeCAD/releases/tag/2022.10.21-2-edge) which was released yesterday doesn't have "Stable" assets, only "Daily". I maintain an AUR package for stable AppImages of FreeCAD link (https://aur.archlinux.org/packages/freecad-assembly3-appimage), therefore I'm unsure what I should do with this latest release. Is it planed to publish a "Stable" AppImage asset on this release? If not, would it be possible to mark this release as a pre-release? Is the release process documented somewhere? As it seems "Daily" AppImages aren't actually built daily, how Daily and Stable AppImages differ?

Thanks!

realthunder, 2022-10-23:
The `Daily` branch is forked from `Stable` when I did a major update of the topo naming encoding, and then a merge with upstream 0.19. And now it is time for a big merge with 0.20, so the `Stable` branch is about to retire.

Please note that `Daily` branch build is supposed to open document authored in `Stable` without any problem. It is not so the other way round.

gilbsgilbs, 2022-10-24:
Thanks @realthunder . I'll package Daily releases then.

closed 2022-10-24 by gilbsgilbs