realthunder/FreeCAD #605: Can't revolve around sketch line

m-sundman, 2022-10-23:
Normally in the PartDesign Revolution tool you can select a line from a sketch as the axis, but now that makes it revolve around some unpredictable axis instead. (As a workaround you can make a Datum Line from the line in the sketch and then revolve around that datum line instead, and that works as it should.)

OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.1021.26244 +5249 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d307b5c936403e8dca3f24b5c1e64bd16c88b28d
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

pathfinder555, 2022-10-31:
Can confirm, Revolution seems broken.

realthunder, 2022-11-01:
Should be fixed by 3c8273fb8c5. You can test snap build, or wait for the new release. Feel free to reopen if still have problem.

closed 2022-11-01 by realthunder