realthunder/FreeCAD #611: TechDraw Issue in RealThunder

richwein3, 2022-10-26:
I have an issue with TechDraw. When I create a new view on a new sheet, everything looks fine. But after I save it and exit realthunder, and then load the project again, the dimensions are separated from the view drawing, always down and to the left.

To duplicate this I can draw a circle, pad it, create a view on the sheet, save,close and reopen.

I discovered by hovering over the Page - Template property that it can't find the /tmp/... template, in red in the image below. Maybe I have some path, or other configuration messed up.

Note: If I open that same file created in RealThunder in the Master or the Topo all of the dimensions are properly aligned with the drawing. So RealThunder did create the TechDraw correctly, it just doesn't read it in correctly

Thanks for help,

Rich

[image]

OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.1025.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.10.25-edge)
Hash: 4caae98c6355bfc2d0fe660321172af073bd5508
Python version: 3.8.10
Qt version: 5.15.5
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

richwein3, 2022-10-26:
Some additional information.

I now ran RealThunder from the terminal - reloading a simple cylinder previously created, dimensioned and saved, using the default A4 template.  It looks like it is not finding some of the files in the snap install.  I had used snap remove and a reinstall several days ago, and the recent update.
--------------
rich@rich-desktop:~$ freecad-realthunder
FreeCAD 0.20, Libs: 2022.1025R14555 (Git shallow)
Adding snap-specific PYTHONPATH to sys.path: /home/rich/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/87/lib/python3.8/site-packages:/snap/freecad-realthunder/87/usr/lib/python3/dist-packages
connect failed: No such file or directory
Template File: /snap/freecad-realthunder/82/usr/share/Mod/TechDraw/Templates/A4_LandscapeTD.svg is not readable
Line Group File: /snap/freecad-realthunder/87/usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
Line Group File: /snap/freecad-realthunder/87/usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
Line Group File: /snap/freecad-realthunder/87/usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable

[image]

Neubo, 2022-11-02:
Hello,

I have the same issue :(

richwein3, 2022-11-02:
Select the TechDraw workbench first or techdraw preference section may not appear, then go to preferences, techdraw.  Files - Default Template Line Group File.
On my system it was:  /snap/freecad-realthunder/89/usr/share/Mod/TechDraw/LineGroup/LineGroup.csv
Note that path starts at root.  My templates start at /home/myname/snap/freecad-realthunder/...

Just in case its not exactly the same as my system, search that snap path to make sure it is there, then paste it.  I hope it works for you.

I assume it should have initialized itself during the snap install, but didn't.  

Now the offset problem unsolved, which works fine in the topo releases but not link.

richwein3, 2022-11-02:
I hae the following terminal window text on load, see below.  The Hatch, Welding, and Pa Hatch are missing, probably fix same way, probably off root.  Check the Topo or Master to see what is in preferences.  I just didn't bother to fix those others yet.  'is not readable' means 'file not found' I think since the path isn't in preferences.

FreeCAD 0.20, Libs: 2022.1031R14555 (Git shallow)
Adding snap-specific PYTHONPATH to sys.path: /home/rich/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/90/lib/python3.8/site-packages:/snap/freecad-realthunder/90/usr/lib/python3/dist-packages
connect failed: No such file or directory
Svg Hatch File:  is not readable
Welding Directory:  is not readable
Pat Hatch File:  is not readable

richwein3, 2022-11-02:
I work around the problem also.  Its OK.

Note in the techdraw preference path names contain the 87, 89, and latest is 90.  This number is related to the RT snap updates, and should be replaced by '**current**'.   In the terminal window, you see the 90 PATH when you execute RT.

When I launch RT from the terminal window, after the PYTHONPATH is emitted, you will see:

**connect failed: No such file or directory**

That message does not emit until after 'initializing' message goes away as RT loads.  I have no idea what file or directory is not being found, and I don't know what connect failed means.  

RT Techdraw used to align dimensions with the techdraw views just fine several versions ago.

I'll leave this open for now.  I didn't know if anyone else had the same problem.  So thanks for that.  My guess is that most people don't have this problem, which probably means I have something misconfigured.

Neubo, 2022-11-03:
hum, I didn't understood your work around about the offset issue.
I run the Windows version of RT, neither the snap nor in command line :)

But I agree, the workarounds should be implemented directly.

richwein3, 2022-11-03:
TechDraw Bug: The dimensions are not attached or coupled to the view **after reloading**.  When you move the graphic, the dimensions do not follow.
  
New Session: Create a cylinder 1" diameter
-- Create techdraw TOP view, dimension circle diameter = 1"
-- Drag the view, the dimension follows ( works correctly )
-- Save
-- Reload same file
-- Drag view, dimensions do not follow (it also have a crazy zoom initially)
-- Set view property X,Y to 0,0 and the dimension is aligned with the graphic, the dimensions are stuck at 0,0
-- So, the dimension lines are not using the X,Y properties as they should be

This bug is in the RT code, since Topo and Master are able to properly move views created in RT.  

BTW my workaround is to not use RT for TechDraw work.  Switching between RT and other versions might create file compatibility problems though. 

Nuebo, I'll post this more concise description in a new post, if no one else looks at this.  Let me know if you have the same conclusion.  Its a bug, maybe a bug that just a few people have for some reason.

richwein3, 2022-11-05:
Version 2022.704 TechDraw works, so I went back to that as my workaround.

closed 2022-11-05 by richwein3

Neubo, 2022-11-07:
Hello @richwein3 ,

yes, I agree. Let's create a new post with the offset issue.
Thanks !