realthunder/FreeCAD #612: [Feature idea] cutting bodies by open cotour operations

oscilococcinum, 2022-10-26:
Hello

Currently PartDesign Pocket and Groove don't work with open contours. That's actually very convenient since PartDesign, as far as  I know, is a solid oriented workbench. However I see application for features made with open contours. In LinkBranch already exist something similar in its philosophy, the "Slice" feature. Idea its kind of fusion of that two tools (for example "pocket" and "slice").

Let's say we have a body that we want to cut on one side. The easiest way to do so its to draw a closed contour in sketcher and use pocket feature. 
[image]
[image]
Now lets say we incorporate slice tool and extrusion into pocket feature, in a way that pocket feature used on open contour results with cut of body by surface extruded by that operation, and like in slice, the option to select one of the results. 
[image]
[image]

Personally I see that as a massive workflow improvement.

wohltat, 2022-10-26:
It does not need to be a closed contour. But you still need the extrusion step.

OficineRobotica explained it nicely in one of his tutorials:
https://www.youtube.com/watch?v=N7b2LBgREmE&t=1237s

oscilococcinum, 2022-10-27:
Yes that's right. The idea is basically to incorporate split operations into pocket/groove features, in a way that these work like [extrusion/revolve of cutting surface + split]. Like I said before im aware of workflow that gives the same result, however In my opinion that fusion would add much more transparency in tree.

wohltat, 2022-10-27:
Ah ok, i see now what you meant.

But how should this work with open contours. Then there would be no inside and outside and it would be undefined which part to cut away. In other words the "subtractive" vs "intersecting" settings would not make sense then.

Let's say you cut with a straight line. Which side is chosen to remain.

oscilococcinum, 2022-10-27:
I see at least two possible solutions. First one already exist in slice feature (last photo). User can choose which side remains by double click on "Split_i0" or "Split_i1" or another if result of cut produces more solids. Alternatively there always is option to add property in pocket widget like "Other side", but that would limit user to cutting body by contours that results in only two solids. Because of that I personally consider existing solution is more flexible.