realthunder/FreeCAD #613: Thickness

ljungis, 2022-10-27:
Hello,
I am running Freecad link branch vers. 2022.1026 on a Linux mint 20.3, Cinnamon, 64-bit.
I try to make a thickness on a pretty simple body, selecting two faces to open. But only one will be opened.
Why not the small one? 
[image]
[image]

[Thickness_2.zip](https://github.com/realthunder/FreeCAD/files/9882581/Thickness_2.zip)

ljungis, 2022-11-02:
What does the OCC label mean, and how does it affect a  solution?

discip, 2022-11-02:
OCC refers to the kernel.

Look here:

https://www.opencascade.com/open-cascade-technology/

kind regards

ljungis, 2022-11-03:
discip, thank you very much for the information. However it seems to me that a solution for the thickness issue is far away.
Is there any work around?