realthunder/FreeCAD #620: Sometimes FreeCAD makes wrong intersecting geometry in PartDesign workbench

AlexStankov, 2022-11-01:
OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Steps to reproduce:
1. Extrude cylinder in part design workbench:
[image]
2. Extrude another cylinder in part design workbench which intersects first one. Preview looks good:
[image]
But when I press OK button - FreeCAD shows strange intersction:
[image]

Expected behavior: FreeCAD adds second cylinder to the first one.
Observed behavior: FreeCAD creates strange form which is not a solid body.

This behavior have been checked in two latest FreeCAD versions: 22.07.09 and 22.10.21.

realthunder, 2022-11-01:
Hmm... Still can't reproduce it. I got the correct output on both 0709 and 1021 release. Although the chances are slim, could it be any external workbench you've installed?

https://user-images.githubusercontent.com/1207888/199150078-13408570-31fb-49c2-881e-a11678022cf3.mp4

AlexStankov, 2022-11-01:
Hmm. It is strange..
Finally it has been recomputed correctly after several attempts. But here is another strange behavior. I've added another sketch and made a pad:
[image]

Preview has shown correct result:
[image]
But after pressing OK button - I can see only constructed pad, without previously made cylinders:
[image]

What may be wrong here?
Project file is attached: 
[hanger2.zip](https://github.com/realthunder/FreeCAD/files/9907729/hanger2.zip)

AlexStankov, 2022-11-13:
Again, the same issue with the latest version 221112:
Steps to reproduce:
1. Create sketch with circle and line which crosses this circle
2. Enable "Make internals" option for this sketch
3. Pad one of the circle internals
4. Pad another internal of circle
Expected: both pads are created
Actual: we see strange not closed shape instead of pads.
Preview is good:
[image]
But the actual results is not good:
[image]
[image]

I've tried to recompute any object in the tree, but result is the same. Project is attached.

OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.10.6
Qt version: 5.15.4

Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

[bug.zip](https://github.com/realthunder/FreeCAD/files/9998001/bug.zip)