realthunder/FreeCAD #621: Funky rendering of the shadow drawstyle in macOS

notthewave, 2022-11-01:
Screen Recording 2022-11-01 at 09.11.29

version: 2022.10.21
os: macOS 13 ventura
arch: aarch64

modifications: MidnightUI theme (I think the problem was also present on fresh install)

notthewave, 2022-11-01:
cant confirm this on my linux machine

katosabi, 2023-01-18:
I also see the same issue only in shadow drawstyle on the latest build:

OS: macOS 13.1
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)

Baldomo, 2023-08-29:
The issue is still present on the latest tip (Macbook Air M1, arm64 tip)

```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 2023.816.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * freecad.gears 1.0.0
  * fasteners 0.4.56
  * CurvedShapes 1.0.4
  * ExplodedAssembly
  * Assembly4 0.50.2
```

ademlabs, 2023-09-16:
Maybe this is some incompatibility with the rendering library? Instead of doing grayscale shadows it's doing RGB shadows, this seems like maybe parameters being passed in wrong order in some system API on MacOS?