realthunder/FreeCAD #622: Glass WB in 2022.10.21

panzerottto, 2022-11-01:
In the current version 2022.10.21 the Glass workbench does not work. I know it can be replaced by the TreeView in overlay mode, but it's not the same. Has anything been changed?
On the other hand, the TabBar workbench has resumed functioning correctly.

OS: Windows 7 SP 1 (6.1)
Word size of FreeCAD: 64-bit
Version: 2022.1021.26244 +5249 (Git)
Build type: Release
Branch: LinkDaily
Hash: d307b5c936403e8dca3f24b5c1e64bd16c88b28d
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Italian/Italy (it_IT)

panzerottto, 2022-11-01:
I add that it was also reported in the following thread https://forum.freecadweb.org/viewtopic.php?f=44&t=72971

andyylin, 2023-02-01:
If only I had seen this now. Glass is also not working for me. When I press Q+2 keys, the tree view briefly shows up but then disappears again.

CodeWarriorr, 2023-09-05:
That's a bumer, I have the same problem.
Wanted to try LinkBranch and it feels like next level! Sadly, glass add-on stopped working for me, same behavior, appears for a moment and then disappears.
any1 have a solution for this ?

```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 2023.811.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * fasteners.backup1693898099.801079 0.4.60 (Disabled)
  * Glass.backup1693899161.0023181 (Disabled)
  * freecad.gears 1.0.0
  * Glass
  * A2plus 0.4.60n
  * fasteners 0.4.64
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * Assembly4 0.50.2
```

mcxrrp, 2023-09-06:
Try this....
https://github.com/mcxrrp/Glass_new