realthunder/FreeCAD #623: "Make Internals" feature creating broken geometry

herrgerd, 2022-11-01:
Hey Lei,
I wanted to try your newly added feature.
Created two slots and used this sketch to extrude it in both directions of the plane.

When "refine" is active, it works. If not, it creates something like this:

[image]

Cheers, Jan

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.1021.26244 +5249 (Git)
Build type: Release
Branch: LinkDaily
Hash: d307b5c936403e8dca3f24b5c1e64bd16c88b28d
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)

[error.zip](https://github.com/realthunder/FreeCAD/files/9912842/error.zip)

realthunder, 2022-11-09:
The error seems just went away by simply do a recompute. Note that you'll need to recompute the first Pad. Can you remember you steps to reproduce the error?

herrgerd, 2022-11-12:
> The error seems just went away by simply do a recompute. Note that you'll need to recompute the first Pad. Can you remember you steps to reproduce the error?

Tried it, did work but only after manually recomputing each pad. I guess I'll close this thread now.

closed 2022-11-12 by herrgerd