realthunder/FreeCAD #624: STL export

ljungis, 2022-11-03:
I have designed this body and exported it as an .stl file.
However all the slicers I have tried says it is not a valid stl file.
I have tried Cura, Prusa and Ideamaker.
What is wrong?

[Valve_test.zip](https://github.com/realthunder/FreeCAD/files/9928579/Valve_test.zip)
[Valve_test-Body.stl.zip](https://github.com/realthunder/FreeCAD/files/9928580/Valve_test-Body.stl.zip)
[image]

herrgerd, 2022-11-08:
If you mark the last entry in the object tree and then export, it does work. It's a quirk I encountered several times.

[image]

[Valve_test-BodyPad002.zip](https://github.com/realthunder/FreeCAD/files/9965546/Valve_test-BodyPad002.zip)

ljungis, 2022-11-09:
Thank you Jan. This solved it.

discip, 2022-11-09:
@ljungis
Please don't forget to close this issue if solved.

ljungis, 2022-11-10:
Thank's.

closed 2022-11-10 by ljungis