realthunder/FreeCAD #628: Wrong Shapestring text displaying

vovodroid, 2022-11-10:
[image]

The same file created in FC 0.20.1 looks ok:

[image]

Font file included in zip.

[3D.zip](https://github.com/realthunder/FreeCAD/files/9983648/3D.zip)

vovodroid, 2022-11-11:
Actually file created with FC 0.20.1 looks good before recomputation:

[image]

And after recomputation:

[image]

julia-m-stephenson, 2022-12-26:
I thought I'd try the latest Linkstage release. I still get the same issue.

Here's what happens if I open the correctly working file from Linkstage build.

[image]

Here's the error text for easy copy paste 

`12:07:47  ----------------
12:07:47  GuiCommand: ShapeString
12:07:47  Pick ShapeString location point:
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Body
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Sketch
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pad
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pad
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Sketch001
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pocket
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pocket
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#ShapeString
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pad001
12:10:12  <PropShape> PropertyTopoShape.cpp(422): Pending recompute for generating element map: Richards_Halt_0_20_1_Bounding_Box_Method2#Pad001
12:10:12  Traceback (most recent call last):
  File "C:\Program Files\Realthunder\FreeCAD-asm3-Daily-Win-Conda-Py3.10-20221128-x86_64\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'draftviewproviders.view_shapestring'
12:10:21  <App> PropertyLinks.cpp(401): Richards_Halt_0_20_1_Bounding_Box_Method2#Sketch001.Support auto change element reference Richards_Halt_0_20_1_Bounding_Box_Method2#Pad Face6 -> ;#1e;:G0;XTR;:H911:8,F.Face6
`
So the issue appears to be a missing module 'draftviewproviders.view_shapestring' has this been renamed / refactored in link stage?

I have the file which works okay in vanilla 0.22.1 I'll try and upload it as a test case for you.

Thanks Julia

julia-m-stephenson, 2022-12-26:
Here is my test file hope it helps :-)

[Richards_Halt_0_20_1_Bounding_Box_Method.FCStd.gz](https://github.com/realthunder/FreeCAD/files/10303384/Richards_Halt_0_20_1_Bounding_Box_Method.FCStd.gz)

julia-m-stephenson, 2022-12-26:
Having found where that module is in Vanilla freecad I can see the module is missing from the realthunder branch

C:\Program Files\Realthunder\FreeCAD-asm3-Daily-Win-Conda-Py3.10-20221128-x86_64\Mod\Draft\draftviewproviders

[image]

But it is in the vanilla freecad directory.

C:\Program Files\FreeCAD 0.20\Mod\Draft\draftviewproviders

[image]

Is this just a build issue?

Thanks
Julia

julia-m-stephenson, 2022-12-26:
My bad it looks like it was a bug in the Python 3.10 build for windows only. See this forum post [https://forum.freecadweb.org/viewtopic.php?style=4&f=3&t=71036](url)

Fixed in vanilla Freecad by this commit.
[https://github.com/FreeCAD/FreeCAD/commit/3b29cf18be](url)

commit 28f7f43c (2022-12-26): [Part]fix makeWireString for Win & Py > 3.09

Fixes #628
@julia-m-stephenson
closed 2022-12-26 by realthunder