realthunder/FreeCAD #629: Vertex color not exported to .step file

WinstonMechanics, 2022-11-10:
Hi!

I noticed that face and line color are correctly exported into the .step file but vertex color seems not to be transferred into the .step file.

The exported model should have vertex color as red (255, 0, 0), edge color as green (0, 255, 0) and face color as blue (0, 0, 255). Changing export scheme to AP203, AP214 or AP242 did not seem to have any effect. 

Is there a way to enable export of vertex color?

Freecad version: FreeCAD-asm3-Daily-Conda-Py3-Qt5-20221021-glibc2.12-x86_64.AppImage 

**Body in Part Design:**
[image]
**Exported Body:**
[image]
**Step import and export settings:**
[image]

[stepColorIssue.zip](https://github.com/realthunder/FreeCAD/files/9984111/stepColorIssue.zip)

realthunder, 2022-11-12:
AFAIK, STEP do not support vertex color. Maybe some later version supports it, but OCC has no API to export/import the vertex color.

WinstonMechanics, 2022-11-14:
Thanks for the info! 

I’ve been mostly using color to automate a preprocessing task for FEA/CFD use which works great with models that have boundary conditions set to faces or lines, but occasionaly there’s a need to work with vertices too. E.g set all green vertices as supports and all red vertices as loads for FEA. 

Would there be a way to include some kind of metadata into the step model using OCC if color is currently is not supported for vertices?