realthunder/FreeCAD #631: Fillet feature results in wrong shape

oscilococcinum, 2022-11-11:
[image]
[image]
[image]

[bug.zip](https://github.com/realthunder/FreeCAD/files/9993277/bug.zip)

realthunder, 2022-11-12:
I am afraid you've hit the limit of OCC, the CAD kernel. It just can't handle that edge joint. For my testing, you can't only get somewhat correct result with fillet radius <=3mm. You can get bigger radius if you add more fillets around the joint.

oscilococcinum, 2022-11-22:
I have tried different approach and result is slightly different. From geometry perspective the solid is the same as in earlier loft approach, but i suppose that there are some differences in BREP model. But nonetheless its rather weird that it works almost perfect when I'm not using loft. Maybe it will help someone.

https://user-images.githubusercontent.com/8177
[another.zip](https://github.com/realthunder/FreeCAD/files/10071165/another.zip)
8075/203424065-dc87a17c-d374-43d8-b97a-75b0ce7c727b.mp4

OS: Linux 6.0.9-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.31553 (Git)
Build type: Release
Branch: makepkg
Hash: d18dbf08841c51a61705a591ff87bf465c193495
Python version: 3.10.8
Qt version: 5.15.7
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)