realthunder/FreeCAD #635: Open/Save dialog freezes FreeCAD window for 5..30 seconds

AlexStankov, 2022-11-13:
When I press SaveAs or Open button - FreeCAD freezes for 5...30 seconds. The system window "FreeCAD Link Branch isn't responding" appears:
[image]
This freeze may appear again during surfing files.

May it depend on my OS (Ubuntu 20.04) or any installed libraries/programs? This behavior has not been observed with any other software on this computer. 

This issue is always reproduced by two latest FreeCAD Link Branch versions 221021 and 221112.

OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.10.6
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

m-sundman, 2022-11-14:
Do you have some remote filesystems mounted?

AlexStankov, 2022-11-14:
> Do you have some remote filesystems mounted?

No, I haven't. And why should FreeCAD scan all the mounted FSs? As I already said - FreeCAD is the only program which freezes surfing any file dialogs. And it wasn't reproduced on any version later than 221021.

AlexStankov, 2022-11-15:
> To debug the problem, you could try some other Qt program using the same Qt version.

Which Qt version is used in the latest FreeCAD revisions?

m-sundman, 2022-11-16:
> Which Qt version is used in the latest FreeCAD revisions?

The first line of the release notes says: _"(New) Update conda release for all platforms to Python 3.10, Qt 5.15 and OCC 7.6"_
and according to your post above you're using: _"Qt version: 5.15.4"_

AlexStankov, 2022-12-23:
Finally, I've solved my issue. [This](https://bbs.archlinux.org/viewtopic.php?pid=2061934#p2061934) topic has helped me a lot:

> I have resolved the problem, the file dialog is instantely opening. I just had to delete the file ~/.config/QtProject.conf and the problem was solved.

closed 2022-12-23 by AlexStankov

m-sundman, 2022-12-25:
> [...] I just had to delete the file ~/.config/QtProject.conf and the problem was solved.

Yeah, that file stores your bookmarks and recently used locations, so if one of those is unavailable then I can imagine there could be a delay. (Although, to be fair, if qt-developers are competent such a thing shouldn't happen, because they'd do any such loading in a separate worker-thread.)