realthunder/FreeCAD #636: Relevant performance Issue with last daily 11/12

riparz, 2022-11-14:
With the latest version released on November 12, a serious performance problem emerged on the MacOs system (Monterey 12.6, 16GB ram). 
Freecad literally goes jerky, never happened before. 
As it stands it has become unusable.
Let me know what kind of information you might find useful. Hello

Riparz

realthunder, 2022-11-14:
Can you please paste the FreeCAD about information here? Which build are you using, the BigSur or Conda? Or are both perform the same?

riparz, 2022-11-14:
Here
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git)
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.10.6
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: C/Default (C)

I'm using Conda build.
Let me know if you need other info
thanks a lot
M

realthunder, 2022-11-14:
Can you please try BigSur build and see if it has the same problem?

riparz, 2022-11-14:
> Can you please try BigSur build and see if it has the same problem?

Already done, I edited my previous post. 
With bigger build is the same.
With old build works fine

riparz, 2022-12-02:
hello, just to inform that the performance problem persists even with the latest daily of November 28th and makes freecad practically unusable on my macos. 
I hope it will be fixed soon, for the moment I stay on the latest working version

realthunder, 2022-12-05:
Can you please elaborate on the performance problem? What operation did you perform to trigger the jerkiness? Or does it become slow when it start up?

riparz, 2023-01-08:
I apologize for the delay but I had not seen the request.
freecad loads normally, but as soon as I do any operation, for example sketch with plane selection, everything is very slow! Unusable. I'm upgrading to macos ventura, tomorrow I'll let you know if the problem persists also with ventura os.

riparz, 2023-01-09:
Hello, this morning I tried with the new Ventura os and the problem persists. 
However I have seen that by disabling the use of openGL (Display preferences) the performance problem disappears. 
So, probably, the cause is related to the OpenGL libraries and not freecad. 
I hope it is useful for your checks.
Bye