realthunder/FreeCAD #637: The new Profile property feature does not work

m-sundman, 2022-11-14:
Areas are not selectable:

https://user-images.githubusercontent.com/10699615/201761019-98c22ae9-d93a-40bc-bd8d-6c041325e339.mp4

OS: Ubuntu 22.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.10.6
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

m-sundman, 2022-11-14:
The "Show preview", "Show on top" and "Transparent preview" pad parameter values make no difference.

realthunder, 2022-11-14:
Can you please post your user settings, `Tools -> Edit parameters -> Export`.

m-sundman, 2022-11-15:
[user-settings.txt](https://github.com/realthunder/FreeCAD/files/10008481/user-settings.txt)

realthunder, 2022-11-15:
I see. You've enabled `Bounding Box` selection style, probably unintentionally. This style is currently not supported for internal shape selection. Just disable it and it'll work fine, `View -> Selection options -> Bounding Box`.

m-sundman, 2022-11-16:
Alright. Yeah, I've never even noticed that Selection options submenu before..
After disabling it the profile thing works.

closed 2022-11-16 by m-sundman

m-sundman, 2022-11-16:
For future reference, here is a list of things to try:
1) Enable: Edit -> Preferences -> Sketcher -> Auto generate internal geometries
2) If you just want it for one sketch, or an old sketch, then in that sketch's data properties set Internal Geometry -> Make Internals to "true".
3) Disable: View -> Selection options -> Bounding Box
4) Select the faces of the profile to extrude first, and then click "Profile"