realthunder/FreeCAD #640: <Exception> Missing external geometry reference

wsteffe, 2022-11-18:
Using file:
[CoaxFilt_5_2_comb.zip](https://github.com/realthunder/FreeCAD/files/10040734/CoaxFilt_5_2_comb.zip)

when I recompute Boolean003 in Nody named DIEL_CAVITY I get the following error in report view:

> 12:02:23  <Sketch> SketchObject.cpp(8303): External geometry CoaxFilt_5_2_comb#Sketch020.e1 missing reference: Import006.;g3;SKT;:H28c,E
> 12:02:23  <Sketch> SketchObject.cpp(8303): External geometry CoaxFilt_5_2_comb#Sketch020.e2 missing reference: Import006.;g4;SKT;:H28c,E
> 12:02:23  <Exception> Missing external geometry reference
> 12:02:23  Recompute failed! Please check report view.

But Sketch020 doesn't even exist in the structure tree. Probably it was present in previous version of this document that I have changed.

wsteffe, 2022-11-24:
I have seen that Sketch020  was included in the dependency graph and used by Pad029. But Sketch020 and Pad029 were not visible in the structure tree. So now I have made them visible and deleted them.
The problem is solved but I do still not understand how and why those objects became not visible in the structure tree. I even do not understand for what reason FC makes it possible to have objects which are not visible in any place and lead to a failure in recompute.
It is a very confusing behaviour which I have never seen in any other CAD.

closed 2022-11-25 by wsteffe

realthunder, 2022-11-26:
You must have accidentally hide the object in the tree using the context menu. To reveal all hidden items, you can right click any where in the tree, and select `Show hidden items`, or use shortcut `(T, S)`. Or better, just hit anywhere in the tree view, and `Ctrl + F` and type the name or label to search for the object, then press a few `Enter` to confirm the selection, and it will reveal the object even if it is hidden.

The item hiding function is give user more freedom to organize the tree.