realthunder/FreeCAD #645: Pocket generates odd/wrong geometry after modifying pocket

andyylin, 2022-11-25:
I'm not expert enough to understand the underlying issue here, so in very plain terms the pocket that runs through the hinge used to work fine, but after adding a 0.1mm tolerance the entire geometry goes crazy:

[image]

Here's the original file before changing the tolerance.
[供料機案件蓋子-switch-cover.zip](https://github.com/realthunder/FreeCAD/files/10087869/-switch-cover.zip)

Here's the file after just modifying the pocket tolerance to 0.1mm:
[供料機案件蓋子-switch-cover-error.zip](https://github.com/realthunder/FreeCAD/files/10087870/-switch-cover-error.zip)

Here's the file after trying to change the tolerance back to 0.1mm. The odd geometry remains:
[image]
[供料機案件蓋子-switch-cover-error2.zip](https://github.com/realthunder/FreeCAD/files/10087874/-switch-cover-error2.zip)

----

As another reference, the same file also ran into a pocket issue that originally worked but after modifying a previous sketch, no amount of adjustment would cause the pocket (Pocket006) to work properly again.
[image]
[供料機案件蓋子-switch-cover-error3.zip](https://github.com/realthunder/FreeCAD/files/10087882/-switch-cover-error3.zip)

realthunder, 2022-11-26:
This kind of weird shape usually mean that there is some error in early modeling step. For your case (the first attached file), it is the `PocketBase` as shown below

[image]

I've been seeing this kind of failure popping up more frequently lately. I am not sure what's the actual cause. Please right click the document icon in the tree view and select `Mark to recompute` to mark all objects for recompute. A complete recompute can fix the potential problem in your first file. For your later files, a full recompute is still required. But there will be more errors require manual fixes.

realthunder, 2022-12-10:
Judging by the look of the shape, it indicates that there are errors in your model history. Here is a simple trick to find out which one. Just use the `Check geometry` command button on each early feature. You need select the feature to check first before this button gets activated.

You can trace out that the first feature having some error is `PocketBase`. And it looks like an common OCC (the CAD kernel) tolerance problem. The `UpToFace` option you use produce a cutting tool shape at a critical height which confuses OCC. Using a slight offset to make it just a little bit taller can solve all your later problem.

[image]

[image]

andyylin, 2022-12-12:
Wow, thank you, now I know how to troubleshoot this problem in the future. 

If there was a way to hint to the user what the error is related to, I think it would help more beginners like myself adopt FreeCAD use.

Thanks again, closing the case.

closed 2022-12-12 by andyylin

realthunder, 2022-12-18:
I have added an option to auto validate shape on changes. See [here](https://github.com/realthunder/FreeCAD/issues/662#issuecomment-1356738217) for a demo.