realthunder/FreeCAD #648: Access violation on cancel button

marvinmz, 2022-11-29:
I am consistently getting access violation when pressing the cancel button on the pad and pocket tools in part design.

Steps to recreate:

- in Part Design create a new sketch on the slanted face
- create a rectangle inside the face
- use the pocket tool on the rectangle
- select the cancel button

[dummy.zip](https://github.com/realthunder/FreeCAD/files/10112926/dummy.zip)

realthunder, 2022-11-29:
I can reproduce the problem in 11.12 release, but it seems gone in the latest 11.28 release. Please try and report back.

marvinmz, 2022-11-29:
I am still getting the same error. Is there a way to get a stack trace or something else that would be helpful?

realthunder, 2022-11-29:
It is in user FreeCAD user directory, usually `C:\Users\<username>\AppData\Roaming\FreeCAD\crash.log`. I've seen mine. It crashes in Coin3D. I am not quite sure exactly where nor why.

realthunder, 2022-12-03:
I've found a problem that most likely is the culprit. Feel free to reopen if you still experience any problem.

commit 2a705194 (2022-12-03): Gui: fix accessing deleted object in ExpressionBinding

Fixes #648
@marvinmz
closed 2022-12-03 by realthunder